machinekit/Cetus #2: Some shortcuts does not work

EdgarJRobles, 2016-03-25:
F2,F1, arrow keys can not activate his functionality, I take a look at QML code but it seems syntax does not have issues, It might conflicts with other standart shortcuts.
i'm a newbie of qtquick  xD.
Greetings.

strahlex, 2016-03-25:
Which OS?

EdgarJRobles, 2016-03-25:
Machinekit-client is running under Ubuntu 14.04 3.13.0-83-generic X86_64 and  mkwrapper  is installed on Debian wheezy 3.8.13-xenomai-r78 (BBB).

strahlex, 2016-03-29:
Interesting. Do you use Unity? Maybe the keyboard mapping is already occupied by system keys. I tested with Debian KDE and Windows.

EdgarJRobles, 2016-03-29:
I don't like how unity looks then i change inferface to gnome 3, since i changed to gnome I've had few dependencies conflicts i can't solve.
I have other pc running original ubuntu 14.04 (unity) let me test it and see what happen.

machinekoder, 2017-11-13:
A fix for this problem is coming up shortly.

machinekoder, 2017-11-13:
https://github.com/qtquickvcp/Cetus/commit/a4deac212d8984737c383d0615ae39eb674e8042

closed 2017-11-13 by machinekoder