machinekit/Cetus #8: Adding lathe support

alkabal, 2018-02-07:
Hi

It is a difficult job to add a lathe mode for Cetus ?

I have try to read source code but i have short knowledge with programing, so if you does not want to make the necessary amount of work maybe you can help me for some starting point ?

Best regards, tahnks for your work !

Aurélien

machinekoder, 2018-02-07:
it\s already work in progress and should be available in the next few weeks: https://github.com/machinekit/QtQuickVcp/issues/19

machinekoder, 2018-02-07:
If you want to help please take a look at https://github.com/machinekoder/QtQuickVcp/tree/lathe-support
Preview of G2/G3 moves is still a problem, the rest should already work.

alkabal, 2018-02-07:
Oh this is awesome, i try to read and understand src many thanks !

machinekoder, 2018-02-18:
Lathes will be supported soon: https://github.com/machinekit/QtQuickVcp/pull/264

machinekoder, 2018-02-18:
https://github.com/machinekit/Cetus/pull/9

alkabal, 2018-02-20:
Hi

Many thanks for this commit this is really better now i'm very very happy !

Unfortunly i have allways something to ask lol
It is possible later to add a spindle rpm show box ?
Tool turret direct choose button ?

Thanks for your sharing !

machinekoder, 2018-03-04:
@alkabal Yes, that's possible. Can you please open a new issue for this request.

closed 2018-03-04 by machinekoder