machinekit/Cetus #10: DRO tab not work with lathe

alkabal, 2018-02-28:
Hi

Hi have this problem with your release for lathe, if selected lathe = 1 from file main.ini the standalone tab DRO is not updated anymore. 
Remove lathe = 1 make the DRO tab work fine.

This is not a big deal because from PREVIEW dro showing is ok with lathe mode.

Best regards

machinekoder, 2018-03-04:
Please make you have the most recent versions of QtQuickVcp and machinekit.

Can you post a screenshot?

alkabal, 2018-03-04:
Hi

debian stretch .4.110-ti-rt-r143 #1 SMP PREEMPT RT Fri Jan 19 19:12:19 UTC 2018 armv7l
 machinekit (0.1.1519038517.git3813f0e-1~stretch) 
client 201803031410-master-17a5c9c-x64

Cetus i have  git clone https://github.com/machinekit/Cetus

[image]

Thanks for reply

machinekoder, 2018-03-27:
Please make sure to update your fork.

```bash
add upstream https://github.com/machinekit/Cetus.git
git fetch upstream
git pull upstream/master
```

alkabal, 2018-03-27:
From https://github.com/machinekit/Cetus
 * branch            master     -> FETCH_HEAD
Already up-to-date.

This is same, only preview tab have Rad Dia Z updated position
DRO tab stay X Z and position allways 0

Thanks for help

alkabal, 2019-02-14:
Hi

With fresh instal this problem is not resolved, but finally i think for lathe this tab is not really usefull and can be simply removed.  What do you think about that ?

Br

alkabal, 2019-11-04:
Hi

just for reporting i see the same issue with debian computer. (off course all up to date)

Br