machinekit/Cetus #12: rtapi

alkabal, 2018-03-04:
Hi

It is normal than Cetus don't display this type of message from .comp file ?

I can't debug because SSH and Cetus never display this output.

    rtapi_print_msg(RTAPI_MSG_ERR,"Test message!");

Best regards

machinekoder, 2018-03-27:
Yes, It only displays EMC and NML messages, no RTAPI debug/error output.

machinekoder, 2018-03-27:
That said you can use the `log` service to display the messages. https://github.com/machinekit/QtQuickVcp/blob/master/src/application/applicationlog.cpp

closed 2018-03-27 by machinekoder

alkabal, 2018-03-27:
I'm really sorry bit i don't understand how to use this from toolchanger.comp

                     if(machinehomed && !currenttoolnumber && !bWarn){                                                                          // just warn once
                           bWarn = true;                               
                           rtapi_print_msg(RTAPI_MSG_ERR, "Tool not homed. Use M6Tx to set current tool and do the home sequence\n");                                                                 
                     }

Best regards

ArcEye, 2018-03-27:
Use a system dialog such as whiptail or zenity
```
system("whiptail --title \"ATC not homed\" --msgbox \"Use M6Tx to set current tool and do the home sequence\" 8 78");
```
The toolchanger was written to run on Axis which does display rtapi messages.

alkabal, 2018-03-27:
Is there some include to add ?

i have try some way without good result

For now i haven't any screen connected to the BBB and not really like to add one for use AXIS. (maybee later if really needed)

Thanks

ArcEye, 2018-03-28:
That was just an example of using a dialog program you could use zenity or anything else.
The program needs to be installed.

Yo don't even need a notification, if the lathe is homed and the tool number is not set, the first M6 will set the current tool number to the T number without movement.

machinekoder, 2018-03-28:
@alkabal If you want to show the manual toolchanger dialog (if you really need it) inside Cetus you will need to add a manual toolchanger extension to Cetus itself. Just create a new issue if that's what you are trying to achieve. If you just want to show a message inside Cetus add (DISPLAY, Change me) to your tool change GCode. http://www.machinekit.io/docs/remap/structure/#remapping-toolchange-related-codes-t-m6-m61

ArcEye, 2018-03-28:
He is using an adapted version of an ATC toolchanger component I wrote a while back.

It wrote out using rtapi_print_msg(....) to remind the user to set the current tool when the lathe was homed.  This displays on the status line in Axis.
It is not a remap, so needs to do something via C code from within the component itself to display something in Cetus.

Using a system(..) call was one idea how he could use an underlying dialog prog to do it, depends of course what system he is running it on and what is installed.

alkabal, 2018-03-28:
I have rewrite the toolchanger.comp for better compatibily with my lathe and add some error message. (planned to share on googlegroup if i found a way for error message)

Yes i like the possibility to show message for know what append with component, i think this is a missing function for use full fearture from remote client.

My setup is BBB with furaday cape and i like using this for a lathe with automatic toolchanger, with a pendant but without any screen.

I want to use the lathe standalone with pendant XHC like a conventional lathe 
Also use from android tablet or windows computer. 
I can order a hdmi lcd but my thinking is : this is better to order a tablet or nothing and use Cetus.
For now except message from component Cetus seems to be a very good choice. 

Off course i can understand that is not possible, but i don't like the idea to instal another app on my computer ony for error message, else over SSH i never see this rtapi message.

Other issue i don't found a way for go outside of error without restarting config
net a-tool-estop axis.3.amp-fault-in <= toolchanger.outestop

Really sorry for my short knowledge

[toolchanger.comp.txt](https://github.com/machinekit/Cetus/files/1857372/toolchanger.comp.txt)

``

alkabal, 2018-05-03:
Hi all

Do you if Gmoccapy is able to display error message from "rtapi_print_msg" ?

Or have you a way for help me to start writing a better toolchanger compatible with all GUI and client ?

Best regards

alkabal, 2019-01-10:
issue found...

alkabal, 2019-01-11:
finally not found  

i'm 99% sure with yesterday i have see this message one time, only one time impossible to make working again.

 rtapi_print_msg(RTAPI_MSG_DBG, "Tool not homed. Use M6Tx to set current tool and do the home sequence\n");