machinekit/Cetus #19: Touch off Lathe mode

alkabal, 2018-09-29:
Hi

When try to use Touch off button with X work fine

Try with Z don't work, error reported Bad character 'u' used

Br

alkabal, 2018-09-29:
It is possible to add automatique G43 after editing tooltable ?

alkabal, 2019-02-14:
Hi

I have try with german language the issue is same.

emc/task/emctask.cc 399: interp_error: Bad character 'u' used
Bad character 'u' used

I have now see something in the dialog box

when X selected "Enter X coordinate relative to workpiece"
when Z selected "Enter undefined coordinate relative to workpiece"

[image]
[image]

I have check source code from Cetus and QtQuickVcp but i don't found nothing than i can understand related to this with wy short knowledge.

I hope this is not a personal config related problem.

Br

luminize, 2019-02-14:
Just thinking out loud:

Try switching to english (keyboard) and see if the problem is still there. You might think you’re pressing a correct key, which might be sending a wrong key.

alkabal, 2019-02-14:
Hi

Many thanks for reply

i think the word Bad character 'u' used is not keyboard related because i does not use the keyboard at all for try to touch off, this is i think because the variable contening the axis selected is badly initialised with Z in lathe mode and the u is the first caracter from word undefined.

but off course i will try you suggest these evening and at same time try from linux computer.

Br

alkabal, 2019-05-06:
Hi

Sorry 3 month later...

I have the from Linux computer but the issue is same.

Br

I have try from Linux computer

alkabal, 2019-11-04:
Hi

Nobody have the same problem ?

Br

alkabal, 2020-01-05:
Hi

Regarding to this problem i have today see something.

Adding a W axis to my config and now : 
when Z selected "Enter W coordinate relative to workpiece"
when W selected "Enter undefined coordinate relative to workpiece"

Br

alkabal, 2020-01-05:
My config are without Y axis so i suspect something like a bad increment for selecting the actual axis for display and use the dialogue box.

Br