machinekit/Cetus #21: Joint following error on larger file

pauloasherring, 2018-10-02:
Hi there! I most certainly don't know this is the right place for it, if it isn't, I apologize in advance.
I am running machinekit remote, mkwrapper as DISPLAY, along with cetus interface as remote interface, on a BeagleBone Green, rt-preempt on a 4.4 kernel, most recent (as of 10/2/2018) apt-get available machinekit release, latest machinekit client release.
I am having issue with joint following errors.
It happens if I go any further on a given long ngc file. If I reduce a lot the feed rate (from 1800 mm/min to 600 mm/m), it doesn't happen.
Tried varying servo thread period and reducing stepgen period, without success.
If I execute the exact same a remote X session over ssh and the axis front end, I get no problems.
Is there a way for me to debug this issue any further?
Thanks!
Paulo Sherring.

machinekoder, 2018-10-07:
Can you please try to monitor the CPU load using top or htop. Is there any significant difference between Axis and mkwrapper? Also please make sure `CYCLE_TIME` time is equal in both setups.

presslab-us, 2019-01-04:
I'm also seeing joint following errors when using Cetus & mkwrapper, but file size does not seem to matter.  Axis works fine, I have run a few programs with that with no issues.  In my case CYCLE_TIME is the same.  When I get a chance later I will report on CPU usage.

scottnortman, 2019-01-16:
I am running BBB+CRAMPS+machinekit 2018-12-10 and am having strange following errors when I use the mkwrapper as a display.  I can run the same file with ssh + X using AXIS gui with no issue.

I am also getting unexpected behavior of the limit switches, where the Cetus GUI indicates they are asserted when the are not.

When I monitor the CPU with htop, soon after I start execution of my *.ngc file, the CPU load spikes to 100% and I get following errors.  The following errors have happened on all axes (X,Y,Z) on different trials.  This is a repeatable issue.

Also, I am using open loop steppers so the following error is confusing to me...

Please let me know if I can help debug.

Scott

pauloasherring, 2019-03-17:
Hey there! I was finally able to get back at this. I also found the CPU spike after the "play" button is pressed and I still have the same issue: with longer files and higher speeds, I get the joint following issue. 
I really would like to get this to work as I don't like at all using AXIS over SSH :(
If there is anything you would like me to try, please advise :)
Paulo Sherring

pauloasherring, 2019-03-20:
Hey, I guess I found the offending stuff! Try this out:
On your python starting script, try changing the mklauncher debug level from 5 to 1:
launcher.set_debug_level(5) ---
launcher.set_debug_level(1) +++

I found that journald spiked hard when debug level was high. It all works here! :D
It is very likely that this issue is platform dependent, since BBB uses slow access memory (i.e. sd card or even embedded flash).
@presslab-us and @scottnortman : if you guys could try this out, it would be great to have a more consistent solution :)
Paulo Sherring.